darthcloud/BlueRetro #1102: GameCube Blueretro conflicts with Wavebird

Aleksi87, 2024-10-17:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

Myself

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Wavebird

### What is problem? (only list ONE problem per report)

After I soldered the ESP32 HW1 on my Cube's controller board it works as it should with modern BT controllers but strangely enough the Wavebird controller stopped functioning since then. The analog stick does not work at all, I tried different frequencies like 1,6,8,16 but the issue persists.
When I plug in a wired OEM controller it works perfectly fine.

I also tried to add a small switch on the negative of ESP32 so that I can turn it OFF when I want to use Wavebird but when I do turn off the ESP32 the controller board does not work at all, it does not recognize any controllers at all!

If ESP32 can not coexist with Wavebird in peace thenn please provide instructions on how to disable ESP32 before using Wavebird.

### What did you expect to happen?

Wavebird should function properly with ESP32 Blueretro

bkacjios, 2024-10-24:
Sounds a little similar to the issue I put in #1088. I have the full on laserbear kit though.

Aleksi87, 2024-10-25:
Edit: I have followed the MundoYakara method actually, maybe that is the reason I have these issues?
https://www.youtube.com/watch?v=BrarEgOK4uw&t=1057s&pp=ygUVbXVuZG8geWFrYXJhIGdhbWVjdWJl

darthcloud, 2025-04-19:
You can't make an internal mod this way.

You basicly got two devices talking on the same port simultenously. This will create issue.

You need something more complex like the lazerbear mod for this to work.

closed 2025-04-19 by darthcloud

Aleksi87, 2025-04-28:
Thank you very much for the response darthcloud! Please could you provide links to the individual pcbs and components each of them need to achieve an internal Blueretro result without having to order the Laserbear Controller PCB from USA ? You see I live in Albania and ordering from USA is very expensive and time waiting.

Thank you in advance!