darthcloud/BlueRetro #1103: Random false joystick inputs

GlydetheSurfer, 2024-10-17:
### BlueRetro firmware version

ngcblueretro 4controllers [Based on blueretro 1.9.2]

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

RetroScaler Official Store (from aliexpress)

### System used

Nintendo GameCube

### Bluetooth controller brand & name

PS3 Sixaxis

### What is problem? (only list ONE problem per report)

False inputs with joysticks while playing games, Both c stick and left stick would jerk diagonally to the top left for no reason without me doing anything.

### What did you expect to happen?

No random false joystick inputs...

https://github.com/user-attachments/assets/317743be-2670-4817-9c26-8a948ab6272b

darthcloud, 2024-10-17:
Update to v24.10 that got fixed

Dup #1021

darthcloud, 2024-10-17:
The firmware is the same for everything.

If the LEDs don't work that tell me you use a Chinese adapter that need a custom FW.
Ask the person you gave money too for support. Sorry.

closed 2024-10-17 by darthcloud

GlydetheSurfer, 2024-10-17:
> The firmware is the same for everything.
> 
> If the LEDs don't work that tell me you use a Chinese adapter that need a custom FW. Ask the person you gave money to for support. Sorry.

which firmware should i use im using a ngc 4 plug adapter
[image]

GlydetheSurfer, 2024-10-17:
i fixed it.. turns out i had to factory reset device XD