darthcloud/BlueRetro #1110: HP 975 Dual-Mode Wireless Keyboard not working with Dreamcast

LightTreason, 2024-11-03:
### BlueRetro firmware version

v24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

RetroScaler

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

HP 975 Dual-Mode Wireless Keyboard

### What is problem? (only list ONE problem per report)

I'm trying to use this keyboard: https://h20195.www2.hp.com/v2/GetDocument.aspx?docname=4aa8-0290enw. It pairs fine to the BlueRetro dongle, but The Typing of the Dead is stuck at the first screen asking me to plug in a keyboard.

I already set the matching Output (Output 1) mode to Keyboard. The Dreamcast does see the connected controller in the system menu because it shows the virtual memory card associated with it.

I'm attaching a Wireshark trace while connecting the keyboard to my Windows device. During the trace, I paired it and the typed `1`, `2`, `3`, `4`, `5`, `6`, `7`, `8`, `9`, `0`, and `Enter`.

### What did you expect to happen?

I hoped the game would recognize the keyboard and work.

### Attach files like logs or Bluetooth traces here

[hp keyboard trace.zip](https://github.com/user-attachments/files/17612431/hp.keyboard.trace.zip)

darthcloud, 2025-01-24:
I made a new way to debug issue like this in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01), could you try to get me a debug trace following theses instructions?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

Thanks

LightTreason, 2025-01-25:
I updated to v25.01 and collected a trace.

Afterward, I tried the game since I hadn't on v25.01 yet. Much to my surprise, it worked! But after ~5 minutes of playing, the game stated that the controller had been disconnected.

I've power cycled everything and it's now back to the state where the game refuses to acknowledge there's a keyboard attached.

I'm not sure what to make of this situation.

Here's the trace in case it reveals something interesting:
[br_debug_trace.zip](https://github.com/user-attachments/files/18545224/br_debug_trace.zip)

During the trace, I paired the keyboard and then typed in "1234567890qwertyuiopasdfghjkl`<enter>`".

LightTreason, 2025-01-25:
Also, I should note that I have a GDEMU with openMenu installed. The keyboard works fine in openMenu (I can select a game and hit 'e' to start it), but the game doesn't detect a keyboard as being connected to the port.

That seems to imply BlueRetro is detecting the keyboard and input fine, but somehow isn't presenting the keyboard in a way the game likes. If there's something I can do to help collect tracing for that, please let me know.

darthcloud, 2025-01-28:
You mention the main menu see the VMU while you are in keyboard mode??

I think keyboard dont have a vmu slot so maybe the games doesn't like that

Try change accessories to none in output conifg

darthcloud, 2025-04-16:
Feel free to reopen if this didn't work.

closed 2025-04-16 by darthcloud