darthcloud/BlueRetro #1118: Ps1 games not getting vibration (Mister FPGA/PS2)

trunk70, 2024-11-16:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 1

### Bluetooth controller brand & name

RetroScaler-Mini adaptateur 

### What is problem? (only list ONE problem per report)

No vibration in PS1 games at all (i play on Mister FPGA and i have tried on PS2 too). With a wired gamepad, all is ok on Mister and PS2 with PS1 games and Vibration ok with PS2 games on PS2 system. Only a problem with PS1 games vibration. 

### What did you expect to happen?

Vibration works in PS1 games :-)

darthcloud, 2024-11-16:
Did you enable rumble to begin with?
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#26---example-enable-rumble

trunk70, 2024-11-16:
Yes Gamepad and Rumble are "on". On PS2 with PS2 games absolutely no problem with vibrations and my blueretro, only with PS1 games. And if y try with my GPadvance adapter on PC, rumble test, all is ok. As if it were in playstation 2 mode only (for the vibrations)...

trunk70, 2024-11-16:
I just did a test on PS1 and it worked. So the problem is on Mister FPGA with my GPadvance adapter... I will do another test on PS2.

trunk70, 2024-11-16:
Ok, so after many tests, it seems that the problem is the combo GPadvance from bliss box (up do date with last formware) and the Blueretro (up to date). Works very well with wired Dual shock and Dual shock2 + gpadvance on mister fpga. Bluertro works on real hardware PS1. Problem is GPADVANCE + blueretro = No vibration in PS1 games (Mister FPGA and PS2 (to be confirmed for the PS2, DVD player is unreliable for the tests...).

closed 2024-11-16 by trunk70