darthcloud/BlueRetro #1119: Dual Joycon mapping

Florin9doi, 2024-11-17:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

-

### System used

Sony PlayStation 2

### What is problem? (only list ONE problem per report)

- The pairing guide links to a non-existent page: https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#63---dual-joycon-pairing-guide
```diff
- https://blueretro.io/blueretro_presets.html
+ https://blueretro.io/presets.html
```
- L1 and L2 buttons are reversed in "Switch Left Joycon Upright": https://github.com/darthcloud/BlueRetroWebCfg/blob/3d54f2e03dc0480f8430ce063ee67d53e5472e14/map/sw_left_joycon_upright.json#L16-L17
```diff
-        ["PAD_LS",       "PAD_LM",       0, 100, 50, 135, 0, 0, 0],
-        ["PAD_RS",       "PAD_LS",       0, 100, 50, 135, 0, 0, 0],
+        ["PAD_LS",       "PAD_LS",       0, 100, 50, 135, 0, 0, 0],
+        ["PAD_RS",       "PAD_LM",       0, 100, 50, 135, 0, 0, 0],
```

### What did you expect to happen?

-

Florin9doi, 2025-01-24:
Mode incorrect links: 
- https://github.com/darthcloud/BlueRetro/wiki/Debug-trace#step-1---enable-debug-mode
- https://github.com/darthcloud/BlueRetro/wiki/Debug-trace#step-3---download-debug-trace
  - "Visit blueretro.io and click ..." goes to a non-existent wiki page

darthcloud, 2025-01-24:
Thanks for reporting fixed all 3 issues.

closed 2025-01-24 by darthcloud