darthcloud/BlueRetro #1121: 8BitDo N64 Mod Kit Rumble Support

FairyKid64, 2024-11-23:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

BlueRetro

### System used

Nintendo 64

### Bluetooth controller brand & name

8BitDo N64 Mod Kit

### What is problem? (only list ONE problem per report)

Rumble does not work.

### What did you expect to happen?

I expect rumble to work in game that support it.

### Attach files like logs or Bluetooth traces here

Check out about half way through the following video I made for a visual description of the bug:
https://youtu.be/XwiVUPwzmr4?si=IBB3R0Rb5M4RPgYr

darthcloud, 2025-01-24:
Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

closed 2025-01-24 by darthcloud

CaLi-0riginaL, 2025-02-04:
Is there a reason why the rumble is so weak for the 8bitdo mod kit ? Is it just the hardware?  I noticed that the NSO controller's rumble is way more intense.

FairyKid64, 2025-08-09:
Nice!  It is fixed!  Thank you!!