darthcloud/BlueRetro #1122: 8BitDo N64 Mod Kit Incorrect Button Mapping

FairyKid64, 2024-11-23:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Nintendo 64

### Bluetooth controller brand & name

8BitDo N64 Mod Kit

### What is problem? (only list ONE problem per report)

When you first sync the controller, the buttons are mapped correctly.  However, if you turn off the controller and reconnect it again without performing a fresh sync, the buttons are mapped incorrectly.

### What did you expect to happen?

I expect the buttons to be mapped correctly when reconnecting the controller without a fresh sync.

### Attach files like logs or Bluetooth traces here

Check out about half way through the following video I made for a visual description of the bug:
https://youtu.be/XwiVUPwzmr4?si=IBB3R0Rb5M4RPgYr

vrmax81, 2025-01-11:
I also experience this. Now one thing that is new for me now is that i plug in a 2nd controller and player 1 is good but player 2 is messed up even if i turn off and on the console.

darthcloud, 2025-01-24:
Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

closed 2025-01-24 by darthcloud

FairyKid64, 2025-02-28:
> Fixed in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)

Awesome!  I can confirm - it is now working!  I made a video testing it:
https://youtu.be/MczQFeUjoM8?si=x6w-WbLDND4609aG