darthcloud/BlueRetro #1125: Game: Golden Axe - The dual (jap) doesn't work.

mhmetalheart, 2024-11-26:
### BlueRetro firmware version

v24.10_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

humble bazooka

### System used

SEGA Saturn

### Bluetooth controller brand & name

8bitdo, Saturn DIY board M30

### What is problem? (only list ONE problem per report)

Game: Golden Axe - The dual (jap), game will be automatically restart. Select modes: game pad: DISC or Saroo ODE or Satiator will reboot again. 
Select modes: gamepadalt. Game can start, but controller doesn't work. 
So, I change back to my original controllers
The 6B pad original: work.
The NiGHT pad: if in 3D: will start the game, but contoller doesn't work. 
Then, I change to Dpad. game reboot. 
So, that means the blueretro gamepad=3D pad. And always in NiGHT controller, and start with Dpad model. Alt=NiGHT controller, but in 3D pad model. Hope you can understand.

### What did you expect to happen?

Should be work.

DarthMotzkus, 2024-11-28:
https://github.com/darthcloud/BlueRetro/issues/995

darthcloud, 2025-01-24:
Duplicate of #995

closed 2025-01-24 by darthcloud