darthcloud/BlueRetro #1128: Updating firmware on Linux - commands not working

mzryz, 2024-12-02:
### BlueRetro firmware version

unknown

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

Blueretro Core

### System used

SEGA Saturn

### Bluetooth controller brand & name

n/a

### What is problem? (only list ONE problem per report)

System: Arch linux (latest, rolling) esptool installed via repos.
1. Extracted firmware (HW1)
2. Entered directory
3. Copy / Pasted the code from 'project documentation / usb serial flash' instructions, code listed on website is: ```esptool.py -p /dev/ttyUSB0 -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash\
--flash_mode dio --flash_size detect --flash_freq 40m 0x1000 bootloader.bin 0x8000 partition-table.bin 0xd000 ota_data_initial.bin 0x10000 BlueRetro.bin```
4. Terminal output was: ```esptool: error: argument operation: invalid choice: 'write_flash--flash_mode' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'read_flash_sfdp', 'merge_bin', 'get_security_info', 'version')```
5. I assumed the syntax was incorrect, tried removing the backslash after 'write_flash' and re-entered into the terminal, but this time output was: ```esptool write_flash: error: argument <address> <filename>: [Errno 2] No such file or directory: 'bootloader.bin```
6. I have noticed the required files are in different directories: Am I supposed to move all these files into one directory? Are the website instructions incomplete? 

### What did you expect to happen?

The firmware to flash

### Attach files like logs or Bluetooth traces here

n/a

mzryz, 2024-12-08:
I'm going to close this issue because I updated by the web GUI. But I think the Linux instructions need updating with more detail because they assume that the user has experience flashing esp32 modules etc. the instructions should be written step by step, not missing anything out.

darthcloud, 2025-01-25:
I made a script to make this easier and updated the wiki
https://github.com/darthcloud/BlueRetro/wiki#82---via-usb-serial

closed 2025-01-25 by darthcloud