darthcloud/BlueRetro #1129: GDEmu black screen when booting with Blueretro

TDM666, 2024-12-05:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Blue-retro (Ali Express)

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

Retrobit Saturn bluetooth

### What is problem? (only list ONE problem per report)

Hi,

I've issue with GDEmu menu (latest firmware) when my blueretro device is plugged.
DC is booting and then giving me a black screen while loading the GDEmu menu.
If I disconnect the blueretro dongle and plug it after GDEMu menu loaded, it's working fine.

Thanks in advance,
Regards.

### What did you expect to happen?

Not freezing

Narcotic82, 2025-01-05:
Hello, I also have the same problem. Any solution?

hudsonbrendon, 2025-01-09:
I have the same problem, any solution?

jerbearwright, 2025-01-11:
I just encountered this, my solution was to format the built in memory card on the blue retro and now it boots normally

darthcloud, 2025-01-12:
Thank you for giguring this out @jerbearwright 

I gave it a shot and can confirm this is the reason.

There 3 options to avoid this:
* Format BR VMU from the Dreamcast VMU manager in the DC boot menu.
* Format BR VMU from BlueRetro web.
* Wait to enter GDemu before associating controller.

closed 2025-01-12 by darthcloud

TDM666, 2025-01-14:
Hi, thanks but under Universal HW1 version, formatting with DC menu is not working, nor deleting mc.bin in the Blueretro web interface. Should I open a sepate ticket for that issue?
Thanks!

corro23, 2025-04-30:
Thanks for the posts. This solved my problem.
I also had this issue I had to boot without pairing a controller then pair the controller and format the memory card on the blue retro.