darthcloud/BlueRetro #1130: Rumble intensity is sometimes strong, sometimes weak.

362053534, 2024-12-08:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

BlueRetro

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

Mini BlueRetro for PS2/PS1

### What is problem? (only list ONE problem per report)

Rumble intensity is unstable. It is sometimes strong, sometimes weak when do an some action.

### What did you expect to happen?

I use ps4/ps3 controllers. 
In any games when I do an same action, rumble intensity is sometimes strong, sometimes weak.
For exsample, when I shoot the handgun in RE4, every times the Rumble intensity is different.

darthcloud, 2025-01-24:
Maybe better in  [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)?

362053534, 2025-01-25:
> Maybe better in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)?

I tried v25.01 today.  But this issue has not been resolved.

362053534, 2025-03-15:
> Maybe better in [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)?

I noticed that this issue only occurs with brief rumbling sounds.
It looks like the duration of each Rumble is unstable.