darthcloud/BlueRetro #1131: GC BlueRetro adapter not working with Wii U Gamecube usb adapter

Uxzylon, 2024-12-10:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler (Aliexpress)

### System used

Nintendo GameCube

### Bluetooth controller brand & name

8BitDo Mod Kit for Original NGC Controller

### What is problem? (only list ONE problem per report)

Inputs from the connected controller aren't recognized by the Wii U using a gamecube usb adapter and nintendon't, despite it working using an original wired controller

### What did you expect to happen?

The controller inputs to be recognized by the system

shadowofdarkness, 2024-12-31:
Did you plug in both USB ends on the adapter? When I made my Blueretro the first test I did was in one of those adapters and the Blueretro could not even turn on due to lack of power until I plugged the second line in. Even though a single wired controller can work with only one USB connection.

Uxzylon, 2024-12-31:
Yes, both ports are plugged in. The blueretro is powered, i was even able to do a firmware update on it

Uxzylon, 2025-01-02:
Just tried it with an xbox controller, same issue. It connects but no response in game. Maybe an issue with my gamecube usb adapter? It's an aftermarket one

Uxzylon, 2025-01-11:
Also tried the blueretro adapter in a real gamecube and it works. So it doesn't work with the wii u usb adapter only, despite it working with wired controllers. Pretty weird issue

Sowden, 2025-01-15:
Over some time, I did some extensive testing for this type of thing.  I did find a solution, I just needed the right device going in the right controller port.  You can read about it here, hope it helps.

https://gbatemp.net/threads/blueretro-for-the-wii-u.618257/

Uxzylon, 2025-02-21:
By replacing my Generic HHC-W201C gamecube usb adapter with a Mayflash gamecube usb adapter, now it works perfectly ! I guess not all generic gamecube usb adapter are made equal

closed 2025-02-21 by Uxzylon

Sowden, 2025-02-22:
Oh now that is interesting.  How far does it get you though, if I may ask?  Are you able to control the Wii U menu screen?  Play any games with it?  Does it mimic as a WUPC?  Because if it does, then I might need to give the Mayflash a try.

Uxzylon, 2025-02-22:
Just like my older usb adapter which was working with wired controllers, it only works in gamecube games using nintendont. Not in the wii u menu or the wii menu. It also works for compatible wii u games like smash bros wii u.

Here's the mayflash usb adapter that i bought : https://a.aliexpress.com/_EuTzQhE