darthcloud/BlueRetro #1135: BlueRetro Dreamcast - 8bitdo Ultimate 2C Problem

jami3rez, 2024-12-27:
### BlueRetro firmware version

24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

8bitmods

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

8bitdo Ultimate 2C

### What is problem? (only list ONE problem per report)

The Ultimate 2C controller connects but behaves erratically, rapidly firing commands I haven't pressed. I have had the same thing happen with two different 2C controllers. The 8Bitdo Arcade stick however behaves completely normally.

### What did you expect to happen?

I expected the controller to not send button presses by itself.

brettinman, 2025-01-06:
Same thing here. I understand the controller does not support x-input over bluetooth so the analog triggers aren't going to work anyway, but I'm a little confused why blueretro seems to be receiving all kinds of phantom inputs when nothing is pressed.

I'm using the LaserBear BlueRetro GC V3 and the [Ultimate 2C Wireless controller](https://www.8bitdo.com/ultimate-2c-wireless-controller/) in bluetooth mode.

darthcloud, 2025-01-24:
Is it better in  [v25.01](https://github.com/darthcloud/BlueRetro/releases/tag/v25.01)?

brettinman, 2025-01-24:
Unfortunately no, same behavior of seemingly random rapid-fire button presses. 

Attaching a debug file where I start the console and connect the controller in question (I had to change the extension to .txt for Github to allow the upload, please change the extension back to .bin).

[br_debug_trace_8bitdoUltimate2C_LaserBearGamecube.txt](https://github.com/user-attachments/files/18541259/br_debug_trace_8bitdoUltimate2C_LaserBearGamecube.txt)

darthcloud, 2025-01-24:
@brettinman your trace is invalid, did you follow the instruction and enable debug mode?
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

brettinman, 2025-01-25:
Ah sorry, I somehow missed that part. Try this one: 

[br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_2.txt](https://github.com/user-attachments/files/18544350/br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_2.txt)

darthcloud, 2025-01-26:
@brettinman 
Can you try this firmware?
https://github.com/darthcloud/BlueRetro/actions/runs/12978399169/artifacts/2487711873

(you need to be logged into github for it to work)

If it doesn't work take another debug trace with it.

Thanks!

brettinman, 2025-01-26:
Thank for the help debugging. Interestingly, on this test firmware it seems to be injecting fewer random presses, but is still pressing them and (seemingly) not responding to actual inputs.

[br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_3.txt](https://github.com/user-attachments/files/18551873/br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_3.txt)

darthcloud, 2025-01-26:
Your trace show that you are still running v25.01 firmware and not the v25.04-beta

brettinman, 2025-01-26:
Yeah I actually just realized that, the first firmware update apparently didn't "take". 

So good news - it works great! I haven't done a full test yet, but no random button presses and inputs seem to work. I'll do a more thorough test and update this thread soon.

brettinman, 2025-01-26:
Ok [GC Controller Test](https://github.com/corenting/GC-Controller-Test) shows all buttons are good, including analog sticks. 

I can't get rumble to work but I think that's expected; from what I can find, the Ultimate 2C only supports Dinput mode when switched to Bluetooth, and Dinput doesn't support rumble.

Output settings: 

[image]

Thanks again for the help here, I'm glad these can be supported since it's a nice controller and there have been some great sales on them recently.

darthcloud, 2025-01-26:
There is a rumble report I can see.

Can you make a debug trace while trying to trigger the rumble (press -/select should do)

brettinman, 2025-01-27:
Sure, let's try these:

With Accessories set to Rumble: 

[br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_rumble1.txt](https://github.com/user-attachments/files/18552146/br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_rumble1.txt)

With Accessories set to None:

[br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_rumble2.txt](https://github.com/user-attachments/files/18552153/br_debug_trace_8bitdoUltimate2C_LaserBearGamecube_rumble2.txt)

darthcloud, 2025-01-27:
Can you try this one and take a debug trace again:
[v25.04-beta-9-g67a86bef_hw2_gamecube.zip](https://github.com/user-attachments/files/18552434/v25.04-beta-9-g67a86bef_hw2_gamecube.zip)

Thanks!

brettinman, 2025-01-27:
Hmm I've tried updating to that firmware a few times now and always get a disconnection at about 7%.

> Reading App version...
Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
app_ver_is_hw2: false app_name_is_hw2: true
Init Cfg DOM...
new_fw_is_hw2: true
> Bluetooth Device disconnected

darthcloud, 2025-01-27:
When it fail just close your Chrome tab and try again in a new one. It will eventually work.

darthcloud, 2025-01-27:
I think your connection issue are due to having debug mode enable while updating firmware.

Turn debug mode off
perform update
then turn debug mode on again to test firmware

Sorry I haven't though of that yesterday.

darthcloud, 2025-01-27:
leave it at rumble, so I guess that means it didn't work?

brettinman, 2025-01-27:
Nope, no dice when pressing "-" or triggering the rumble in the controller test program.

brettinman, 2025-01-27:
Actually scratch that, I'm not sure debug mode was re-enabled. Here's another with debug mode and Rumble selected: 

[br_debug_trace.txt](https://github.com/user-attachments/files/18565238/br_debug_trace.txt)

darthcloud, 2025-01-27:
Ok I made a mistake in previous one.

Hopefully this one will do:
[v25.04-beta-9-g2ddd6d4f_hw2_gamecube.zip](https://github.com/user-attachments/files/18565328/v25.04-beta-9-g2ddd6d4f_hw2_gamecube.zip)

brettinman, 2025-01-27:
Thanks! Unfortunately still no rumble, debug here: 

[br_debug_trace.txt](https://github.com/user-attachments/files/18565429/br_debug_trace.txt)

darthcloud, 2025-01-27:
Ok too bad I guess it doesn't work over BT

They report a rumble report with ID 5 in their HID descriptor.

That report doesn't exist in their LE attribute but there was another output report with ID 1 in the LE attribute.

However writing to that yield a "Error Code: Invalid Attribute Value Length (0x0d)"

As a last ditch efford you could try to connect it on your PC with BT, and launch Steam controller test:
Settings->Controller->Begin test

There a button to trigger rumble.

brettinman, 2025-01-28:
No button in the Steam controller test for rumble when doing the test with this controller connected via bluetooth. The button does appear when connecting over 2.4Ghz USB dongle. There are a number of threads out there of people not getting rumble in bluetooth mode, but it's unclear whether that's because it's impossible or just that none of the common libraries have figured out how to interface with it.

Anyway, thank you so much for the help getting the controller working!

darthcloud, 2025-01-28:
100% 8bitdo software issue. It's all there but it's not hook up.

Thank for testing and providing traces! This helped a lot!

darthcloud, 2025-01-29:
Here a more complete set off builds with this fix until next official release:
https://github.com/darthcloud/BlueRetro/actions/runs/13023080800

closed 2025-02-17 by darthcloud