darthcloud/BlueRetro #1142: RetroTime N64 BlueRetro Adapter losing connection, will not update.

charlesmconard, 2025-01-05:
### BlueRetro firmware version

v24.04_hw1

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroTime

### System used

Nintendo 64

### Bluetooth controller brand & name

Nintendo Switch Pro Controller

### What is problem? (only list ONE problem per report)

It was working fine for the first two days but now the controller won't immediately connect and I have to resync it every time.
I tried to see if a firmware update would fix the problem but I can't even keep a reliable connection, if at all, with the device.

```
Reading App version...
Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
app_ver_is_hw2: false app_name_is_hw2: false
Init Cfg DOM...
new_fw_is_hw2: false
> Bluetooth Device disconnected
```

[image]

### What did you expect to happen?

I can see this screen but it stays at 0% until the connection times out after a minute or two.
[image]

gobbledygoober, 2025-01-10:
I am having similar failed update symptoms. 

Upon receiving the unit, I was able to update (I believe from 1.9-dirty to 24.04), but I have somehow reverted it to the original version it had (perhaps through some button holding). 

Now while attempting an upgrade to 24.10, it fails as described by the OP.

darthcloud, 2025-01-10:
Your best bet on these older FW is to to use a Windows base laptop with Chrome browser. Try multiple time, try to change laptop position around BlueRetro etc.. If nothing work then try another laptop. If that doesn't work try to update it at a different location.

I know it sucks, but there nothing else you can do with BlueRetro that lack a USB serial port.

v24.10 should resolve that problem.

closed 2025-01-10 by darthcloud

gobbledygoober, 2025-01-10:
Happy to report that using a Windows PC worked! 

I was using a Chromebook when I reported the problem above, same 131 major version of Chrome. 

Very interesting, considering I used the Chromebook to update in the past, and the Windows PC is a desktop and much physically farther away. 

Thanks @darthcloud !

charlesmc93, 2025-01-20:
> Your best bet on these older FW is to to use a Windows base laptop with Chrome browser. Try multiple time, use a new Chrome tab for each try, try to change laptop position around BlueRetro etc.. If nothing work then try another laptop. If that doesn't work try to update it at a different location.
> 
> I know it sucks, but there nothing else you can do with BlueRetro that lack a USB serial port.
> 
> v24.10 should resolve that problem.

I do not own any new laptops witn bluetooth capability. I do most of my work from a desktop that has a bluetooth USB dongle. 
When attempting to connect to the adapter I placed that UBS dongle on a powered USB extention cable and had it right next to the adapter. I did not try multiple tabs yet, but I will give it a shot when I have the chance. 

I have a set of these cables on the way so I can backup my original Controller Pak saves while they still work.
https://www.raphnet-tech.com/products/n64_usb_adapter_gen3/index.php
Would it be possible to load the updated firmware with these if the above method fails?

charlesmc93, 2025-01-23:
I tried the process again with the same setup as before. Failed the first time, opened a new tab, connected, chose the firmware update, and let it transfer. 

It just worked this time. I can finally reliably connect to the adapter and I can even read the save files now.
Thank you!