darthcloud/BlueRetro #1155: Blueretro core + Brawler64 doesn't map C buttons

pvgpelvis, 2025-01-27:
### BlueRetro firmware version

v24.10

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Blueretro

### System used

Nintendo Wii Extension

### Bluetooth controller brand & name

Retrofighters Brawler64 Bluetooth NSO Edition

### What is the problem? (only list ONE problem per report)

Hi! I'm having a hard time trying to properly configurate the blueretro core on my Wii, using the brawler64. I have a Wii u pro controller, and it can emulate the wii pro controller perfectly, but when I use the brawler, I can't figurated out how to make all the buttons work as intended. More specific the C buttons don't work all together, in all the presets or the advanced config settings for other controller, always make 2 or 3 work, but never all 4. Am I doing something wrong? I tryied configurate him as a switch N64 controller, tryied switching from Xinput to Dinput, but get the same results.

### What did you expect to happen?

Hope to play the emulator and the n64 virtual console with the brawler64

darthcloud, 2025-01-27:
Update to v25.01 and follow instructions:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#11---retrofighters-brawler64-bluetooth-pairing-guide

Make sure your config is the default one (hold button 10 after upgrading FW to reset config)

darthcloud, 2025-01-27:
Well the c button map to a the wii right joystick so its not possible to press more than two at once.