darthcloud/BlueRetro #1156: Blueretro core (wii)+ Brawler64

pvgpelvis, 2025-01-27:
> Update to v25.01 and follow instructions:
> 
> https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#11---retrofighters-brawler64-bluetooth-pairing-guide
> 
> 
> 
> Make sure your config is the default one (hold button 10 after upgrading FW to reset config) 

 _Originally posted by @darthcloud in [#1155](https://github.com/darthcloud/BlueRetro/issues/1155#issuecomment-2615943501)_

Hey! Sorry for reopening this, I updated the version and reset config to default, the C buttons is all working now. All the buttons seems to be where it should be, except for A and B.

Pressing A, it register as B on wii, and pressing B does nothing.

darthcloud, 2025-01-27:
This is normal you need tochange the mapping:
[image]

darthcloud, 2025-01-27:
btw you can comment on closed bugs dont open a new one.

closed 2025-01-27 by darthcloud