darthcloud/BlueRetro #1157: Does not recognize blueretro in OPL(PS2)

LuxMaximo, 2025-01-27:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter with multiple hardwired cord

### Manufacturer

myself watching youtube videos

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

ESP-WROOM-32

### What is the problem? (only list ONE problem per report)

I tried the new firmware because it said that a new feature was added for PS1, which is vibration for analog sticks. I had the problem that the analog sticks didn't work with my XBOX series S|X joystick and PS4 joystick in the previous version (in OPL games). But they didn't work on the OPL PS2. I tried on my computer and the buttons do work.

### What did you expect to happen?

I was hoping that my analog joysticks would work in OPL PS2 games so I could play with those joysticks that I like.
I will revert to the previous version but I leave the bug in case it helps you.

darthcloud, 2025-05-13:
For me it work but I'm using their old official release from 2021:
sha1sum OPNPS2LD-v1.1.0.ELF
be594ff83a81486c20b3f7276bcef3041bea29f3 OPNPS2LD-v1.1.0.ELF

I can't get the latest betas of OPL to work on my FHDB setup, they just give me a black screen.

One thing you can try is disabling PADEMU feature at both the Global and game level in OPL settings.

darthcloud, 2025-07-13:
Feel free to reopen if this didn't help

closed 2025-07-13 by darthcloud