darthcloud/BlueRetro #1163: Button Combination does not work for Gamecube IGR (in-game-reset) with xbox one/x/s controller

zeus86, 2025-02-09:
### BlueRetro firmware version

v25.01 hw1 gamecube

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

generic  aliexpress one, most likely a clone from retrotime's GC-adapter

### System used

Nintendo GameCube

### Bluetooth controller brand & name

ID 0489:e0e2 Foxconn / Hon Hai Wireless_Device

### What is the problem? (only list ONE problem per report)

I've seen in another issue (#512 ), that it is possible to archieve buttoncombinations by simply assigning a src-button multiple times in a row. 

However, for the ingame-reset of swiss on gamecube this seems not to work. the game registers the buttons in series instead of simultaneously as it seems. The default combination vor an IGR is `RightShoulderButton+Z+B+D-Pad-Down` If i press the button-combination manually, it works, but as a combo it does not. 

I wanted to use the otherwise useless `share` button on the xbox-one/x/s controller, so i removed the default entry for this button, and added the four buttons at the bottom.

Relevant config: 

[image]

(I also tried `Analog R` instead of `Digital R`. Tested with v24.06 (IIRC) and current V25.01)

Am I missing something here?

### What did you expect to happen?

Button Combinations should work at the same time, not serialized.

zeus86, 2025-02-11:
> Did you try to add a 5th entry with Analog R?
> 
> Maybe swiss expect both the analog portion and digital end to be pressed simultanously like a real controller would do.

you are absolutely right, that indeed did the trick, analog r followed by digital r and then the rest indeed worked.
...this would be nice in a default preset, as those modern extra-buttons are unused anyways...

...now i just have to find out, why src- and dst-label  keep resetting on every connect, and all settings seem to reset on power loss... is this normal? or what should I look out for?

however, thanks so far, the initial issue is solved by your suggestion.

darthcloud, 2025-02-11:
label are just to help do the mapping, its not saved

darthcloud, 2025-02-11:
Ok I updated the wiki to clarify this.
https://github.com/darthcloud/BlueRetro/wiki/BlueRetro-BLE-Web-Config-User-Manual#24---mapping-config

Thanks!

closed 2025-02-11 by darthcloud