darthcloud/BlueRetro #1164: N64 Smash Bros Character select bug

SGTgopher, 2025-02-09:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroTime

### System used

Nintendo 64

### Bluetooth controller brand & name

NSO N64 Controller, PS4 Controller

### What is the problem? (only list ONE problem per report)

During character selection in Smash Bros. or Smash Remix, whether using an everdrive or original cartridge, the player character selection will spontaneously reset, clearing your currently selected character and instantly returning the "P1" coin to your floating hand, and returning the floating hand to the bottom left of the screen. It happens often enough that character selection is very difficult and can only be accomplished after multiple attempts and moving on to stage selection faster than the issue can occur. This is the only scenario I see this kind of behavior, and the issue is nonexistent with a standard controller.

### What did you expect to happen?

to be able to select a character normally

darthcloud, 2025-02-09:
Did that adapter ever work? or it's something new after you changed something?

SGTgopher, 2025-02-09:
> Did that adapter ever work? or it's something new after you changed something?

I’ve had the adapter since ‘23 and I’ve always had this issue with every firmware I’ve used

darthcloud, 2025-02-11:
Ok I just tried Smash on my Everdrive and I don't have this issue.

Do you some kind of HDMI mod with a OSD? Those are known to create inteference with BlueRetro.

SGTgopher, 2025-02-16:
> Ok I just tried Smash on my Everdrive and I don't have this issue.
> 
> Do you have some kind of HDMI mod with a OSD? Those are known to create inteference with BlueRetro.

Ahh yes I do. I have the N64 digital hdmi/RGB mod installed. Is that something unlikely to be fixed down the line through a firmware update? (Either blueretro or retro GEM)

darthcloud, 2025-02-16:
See this post
https://github.com/darthcloud/BlueRetro/issues/264#issuecomment-1214224340

You can do as a document it or try get a replacement flex from the hdmi mod vendor.
Or you can simply disconnect the OSD line going to the pif

closed 2025-02-16 by darthcloud