darthcloud/BlueRetro #1170: Ps2 Ps1 memory card load error

unknown2918, 2025-02-13:
### BlueRetro firmware version

v25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

I am

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

dualshock3,4

### What is the problem? (only list ONE problem per report)

My environment
PS2 SCPH-50005 MODBO4.0(modchip)
esp32 internal installation, USB 5v use

I always feel grateful for your hard work.
Thank you.
Updated from v24.10 to v25.01.
After the update, the PS2 browser screen loads the PS1 memory card normally.
However, if you run a PS1 game on a PS2, you will not be able to load the memory card while playing the game.
So I ran a downgrade to v24.10
It works fine in v24.10

### What did you expect to happen?

The same issue is expected to occur on PS1 consoles as well.

darthcloud, 2025-02-17:
I reproduced on a PS1.

Look to be related to some GPIO function rework in esp-idf:
```
e1d3d830ce30e16c6f07236d6ce5b50d76cb251f is the first bad commit
commit e1d3d830ce30e16c6f07236d6ce5b50d76cb251f
Author: Song Ruo Jing <songruojing@espressif.com>
Date:   Wed Aug 28 18:31:32 2024 +0800

    refactor(gpio): public some IO configuration functions
    
    This allows different peripheral drivers to act on the same IO.
```

Looking into it.

darthcloud, 2025-02-17:
This beta should fix it:
https://github.com/darthcloud/BlueRetro/actions/runs/13378257971

egparadigm, 2025-02-26:
Just to say thanks for this. I had a similar problem trying to use Memcard Pro on PS1. The card would not load and showed in xstation as unformatted when using the same port as Blue Retro (the ali express external one). It worked in port 2 but obviously that wasn’t viable. Now working after rolling back firmware.

unknown2918, 2025-02-27:
> 이 베타를 적용하면 문제가 해결될 것입니다: https://github.com/darthcloud/BlueRetro/actions/runs/13378257971

Thank you, it works well