darthcloud/BlueRetro #1172: Blueretro core dumps with 25.01 hw1

asweintraub, 2025-02-16:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

Aliexpress

### System used

Nintendo 64

### Bluetooth controller brand & name

Any

### What is the problem? (only list ONE problem per report)

My blueretro adapter is core-dumping at boot with the latest 25.01 firmware. The issue is the same as in https://github.com/darthcloud/BlueRetro/issues/952, including the same assertion failure:

`assert failed: block_trim_free tlsf_control_functions.h:548 (block_is_free(block) && "block must be free")`

[blueretro logs.txt](https://github.com/user-attachments/files/18813108/blueretro.logs.txt)

### What did you expect to happen?

Blueretro boots.

### Attach BlueRetro debug trace here

See attachment above.

darthcloud, 2025-02-16:
if you upload the n64 specifc v2501 FW does that one work?

asweintraub, 2025-02-16:
It didn't seem to work when I tried it earlier, but I'll try it again get the serial logs in a bit.

asweintraub, 2025-02-16:
Yeah, I tried with the N64 and SNES firmware files and have the same error. The attached file is for N64.

[serial logs.txt](https://github.com/user-attachments/files/18815773/serial.logs.txt)

asweintraub, 2025-02-22:
Any ideas on this? I just tried the latest 25.04 beta and had the same results, so I'm wondering if this could be a hardware issue. I could try ordering another adapter if you think it might help.

darthcloud, 2025-04-19:
Can you try latest beta?
https://github.com/darthcloud/BlueRetro/actions/runs/14544498684

Might be a HW issue no body else reported that and I don't see it myself.

asweintraub, 2025-04-22:
Still no luck with the latest beta on my old device, but I just got a new adapter in the mail today (exact same model from the same seller), and it works just fine with the current release. It definitely seems like I got unlucky with my old hardware, so feel free to close this out.

Thanks for all of your help with debugging this!

closed 2025-04-22 by darthcloud