darthcloud/BlueRetro #1175: [N64] Rumble feedback while toggling between Rumble Pak and Controller Pak is indistinguishable

dagniranum, 2025-02-19:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroTime

### System used

Nintendo 64

### Bluetooth controller brand & name

N64 8BitDo ModKit

### What is the problem? (only list ONE problem per report)

There is a weak rumble feedback while toggling between the Rumble Pak and Controller Pak mode. The feedback is indistinguishable from one another and for the user it is not clear into which mode has been toggled.

### What did you expect to happen?

I expect similar Rumble feedback intensities like the NSO N64 Controller. High intensity rumble when selecting the Rumble Pak and weak intensity when toggling into Controller Pak.

asweintraub, 2025-02-22:
I haven't been able to try the latest firmware, but in the past, rumble mode had a longer rumble, while memory pack mode had a shorter rumble.

dagniranum, 2025-02-23:
@asweintraub I think you are be right with that, nothing has changed with the NSO N64 Controller in the new firmware version, maybe I chose the incorrect words to describe it😄 I corrected the **expected behavior** text to avoid any confusion.