darthcloud/BlueRetro #1193: GameCube Style Controllers Mapped Incorrectly on N64 BlueRetro Adapter

FairyKid64, 2025-03-09:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroSclaer

### System used

Nintendo 64

### Bluetooth controller brand & name

RetroFighters BattlerGC Pro Series

### What is the problem? (only list ONE problem per report)

Hello.  Using GameCube style controllers designed for the Nintendo Switch (like the RetroFighters BattlerGC) with the N64 BlueRetro adapter is challenging because the face buttons are mapped incorrectly - specifically the A and B buttons.

I understand the reason behind the mapping - on a regular Switch Pro Controller, the decision was made to map Switch B to N64 A and Switch Y to N64 B to preserve the layout and ignore the NSO mapping where Switch A is N64 A and Switch B is N64 B.  However, the placement of the face buttons on a GameCube style controller does not lend itself well to this mapping for obvious reasons.

I am aware that this can be configured in the settings and custom mappings can be created, but my problem with this method is that then when I want to use the official NSO N64 controller with the BlueRetro adapter, the buttons are mapped incorrectly, and it's a real hassle to enter settings and remap the buttons every time I switch controllers.

My suggestion is to add an optional feature in the advanced configuration BlueReto page that lets you decide between the B == A && Y == B or A == A && B == B modes.  This would also be nice for players who are not concerned with the original A B button placement, but would rather have the in game prompts match the button names on their Switch controllers, instead of having to remember that B is A and Y is B.

Thanks DarthCloud for all your hard work and letting us submit issues!

### What did you expect to happen?

I would like an option in the advanced config page that allows players to toggle the default B == A && Y == B to A == A && B == B modes that also doesn't mess up the official NSO N64 controller mapping.

Then maybe X could be mapped to C right and Y to C down to make the controls easy for Smash Ultimate players to pick up in this alternate mode.

darthcloud, 2025-04-19:
The real issue is that BlueRetro simply do not have a mapping quirk to "rotate" the button since they use the gamecube layout.

Unfortunatly pad maufacturer almost make it impossible to identify their controller uniquely, so they often just show up as regular pro controllers.

If you can get me a debug trace I might be able to figure something:
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

If the controller has some sort of Xbox/Xinput mode make a 2nd trace in that mode too.

FairyKid64, 2025-04-19:
> The real issue is that BlueRetro simply do not have a mapping quirk to "rotate" the button since they use the gamecube layout.
> 
> Unfortunatly pad maufacturer almost make it impossible to identify their controller uniquely, so they often just show up as regular pro controllers.
> 

Hi darthcloud.  Yes, I agree.  These GameCube controllers certainly would just show up as Pro Controllers.  What do you think of this idea - what if there were an option in the advanced config page to make it so for the N64 adapters, A is always A and B is always B by default for all Nintendo controllers?  This would be similar to how the GameCube BlueRetro adapter works if I'm not mistaken.  Right now on the N64 adapter, for the NSO N64 controllers, A is A and B is B, but for Pro Controllers B is A and Y is B by default.  Could an alternate default mode be added that could be toggled on and off to make all Nintendo controllers act the same way - A==A and B==B?