darthcloud/BlueRetro #1199: Cannot use memory card normally in PS1 games

neoconzer, 2025-03-14:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

DualShock 3

### What is the problem? (only list ONE problem per report)

I bought an BlueRetro for PS1/2 adapter and plugged it into the first controller slot of my fat PS2, successfully connected to the DS3 controller and used normally, then I plugged the PS1 memory card into the first memory card slot, the PS2 can detect the memory card normally, and the saves can be copied or deleted normally in the PS2 browser. But strangely, all PS1 games cannot use memory cards normally. Either the memory card is error or the memory card is not found, and when I replaced the adapter with a wired controller(memory card is still on the first slot) or plugged the memory card into the second slot(adapter is still on the first slot), all games can use the memory card normally(The game needs to support slot 2 memory card).

### What did you expect to happen?

PS1 memory card and BlueRetro adapter can be used normally in all PS1 games in the same slot

fity0612, 2025-03-15:
You can try the beta firmware mentioned in issue #1170 .

neoconzer, 2025-03-17:
> You can try the beta firmware mentioned in issue [#1170](https://github.com/darthcloud/BlueRetro/issues/1170) .

The problem has been resolved, thanks for help!

closed 2025-03-17 by neoconzer