darthcloud/BlueRetro #1200: Xbox Series controller triggers only send full analog range intermittently on GameCube

dstaley, 2025-03-16:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install modkit

### Manufacturer

LaserBear

### System used

Nintendo GameCube

### Bluetooth controller brand & name

Xbox Series Controller

### What is the problem? (only list ONE problem per report)

When my Xbox Series controller connects, it exhibits one of the following states (values are the ones reported by Corenting's GC Controller Test):
- Both triggers work, sending an analog value range of 0-208
- Only the left trigger works, with the right trigger sending an analog value of 0-`some value less than 208`
- Only the right trigger works, with the left trigger sending an analog value of 0-`some value less than 208`

Power cycling the controller will typically put it into another state, with the fully functional state being the most common. Based on my quick testing, it appears that about 20% of the time the controller will be in one of the broken states.

I want to specifically call out that all three behaviors can be observed without any adjustments having been made between them, so this is likely not a configuration issue. This also occurs on multiple Xbox Series controllers, so it's unlikely it's a hardware issue specific to my controller (but it could obviously be an issue with _all_ Xbox Series controllers).

I've attached debug traces for all three states.

### What did you expect to happen?

The controller reliably sends the full range of values for both triggers.

### Attach BlueRetro debug trace here

[three debug traces.zip](https://github.com/user-attachments/files/19274130/three.debug.traces.zip)

darthcloud, 2025-03-16:
1. Is the expected behavior which is the range an official GC controller have.

For the other two make sure you read this from wiki:
https://github.com/darthcloud/BlueRetro/wiki/Controller-pairing-guide#ii---blueretro-joystick-calibration

Basically for Xbox and 8bitdo devices in general you need to press a button like a A a couple time before touching any trigger or joystick otherwise BlueRetro calibration will not be properly set.

dstaley, 2025-03-16:
How long does that calibration persist? Because turning off the controller, turning it back on, pressing A for a few seconds, and then trying the triggers will still result in one of the triggers not working. However if the calibration persists while the adapter remains powered I’d expect that to be the case. Just want to make sure I understand before performing more troubleshooting!

darthcloud, 2025-03-16:
It's not persistant, you need to press a digital button first every time you connect controller.

dstaley, 2025-03-18:
Okay, here's a debug trace where I connected the controller, pressed B and then A a few times, then tried the triggers. The left trigger worked, but the right trigger did not.

Do you happen to know if there are instances where this behavior could be due to the game? For some reason, I have had about 100% success getting the triggers to work when in Swiss and the GC Controller Test app, but this debug trace was while running a game that I have yet to get both the triggers working in correctly. No matter how many times I power cycle the controller (ensuring to press face buttons several times before the triggers), the left trigger always works and the right trigger never does. I thought it might be the game itself, but I tested with an original WaveBird controller and both triggers work as expected.

[br_debug_trace with buttons.bin.zip](https://github.com/user-attachments/files/19304842/br_debug_trace.with.buttons.bin.zip)

darthcloud, 2025-03-18:
which games?

I fixed a bug for Wario world trigger about a month ago

Try this beta:
https://github.com/darthcloud/BlueRetro/actions/runs/13382675228

dstaley, 2025-03-19:
Pokemon XD: Gale of Darkness!

The good news is that I power cycled my Xbox Series controller 10 times with `v25.04-beta-25-g30f2e83` and all 10 had working triggers (as long as I pressed a face button before the trigger). :tada::tada::tada:

I'm going to play around with it a bit more over the next few days and if everything continues to work well I'll close this issue. Thanks for helping me troubleshoot!

closed 2025-03-24 by darthcloud

konwektor, 2025-04-18:
I had similar issues - triggers, and also from time to time some buttons missing inputs.
I use esp32-wroom-32e devkitC module. 
In my case problem was 5V power source for esp32 - floating beetwen 4,6-4,9V.
After providing stable 5V to esp32 issue dissapeared.