darthcloud/BlueRetro #1203: Blueretro internal can´t pair 4 Xbox Series Controllers at the same time

yinxo16, 2025-03-19:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

ESP32

### System used

Nintendo 64

### Bluetooth controller brand & name

Xbox Series S

### What is the problem? (only list ONE problem per report)

I installed an internal ESP32 in my N64. The wiring is correct, checked with a multimeter. When pairing four Xbox Series controllers, right after connecting the fourth controller, all the controllers unpair without exception, regardless of the order in which they are connected. I downgraded to version 1.7 and nothing. I updated to v25.01 and everything is the same. I can use 3 Xbox Series controllers and 1 PS4 controller at the same time without any issue.

### What did you expect to happen?

Pair 4 same brand controllers at the same time.

### Attach BlueRetro debug trace here

[br_debug_trace.zip](https://github.com/user-attachments/files/19346693/br_debug_trace.zip)

darthcloud, 2025-04-18:
Try this beta
https://github.com/darthcloud/BlueRetro/actions/runs/14538838789
login to github to acces the link

I tried with 3 xbox series ctrl + 1 ps4 and it worked fine.

darthcloud, 2025-04-27:
Fixed in v25.04.

closed 2025-04-27 by darthcloud