darthcloud/BlueRetro #1205: Please add support for the 8-Bit do m30 mod kit

charlesmckee47, 2025-03-24:
### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Retro scaler 

### System used

SEGA MegaDrive Genesis

### Bluetooth controller brand & name

8BitDo Mod Kit for Original MD Controller or m30 mod kit 

### What is the problem? (only list ONE problem per report)

Button mapping is extremely messed up X=A
A=B
B=C
C=Z
Z=X 
Y=Y
D-pad up = up
D-pad down = down
D-pad left  = left
D-pad right = right
Start = start 
Mode = mode 

### What did you expect to happen?

All the buttons to map out to what they appear on the controller

### Attach BlueRetro debug trace here

[br_debug_trace.zip](https://github.com/user-attachments/files/19413110/br_debug_trace.zip)

charlesmckee47, 2025-04-17:
Have tried using different modes including x input the switch mode don't remember if I've tried Android mode or mac mode but still none of them have been able to get the button mapping correct

darthcloud, 2025-04-19:
I see nothing in the trace you posted.

Install latest beta
https://github.com/darthcloud/BlueRetro/actions/runs/14544498684

and please redo the debug trace and make sure to follow every step.
https://github.com/darthcloud/BlueRetro/wiki/Debug-trace

charlesmckee47, 2025-04-19:
I will try again to get a debug trace when i get home and text that firmware

charlesmckee47, 2025-04-20:
ok i have tried with the beta and it didn't fix the problem just gave slightly different  resulted hope the debug worked this time as i follow every step on the page telling how to use debug mode
 game pad non alt [3 button mode] 
a=c
c=nothing 
b=b
x= nothing 
y=a 
z= nothing 
start = start 
mode = nothing 
left = left
right = right 
down = down 
up = up
game pad  alt [6 button mode]   
a=c
b=b 
c=z
z=x
y=a
start = start 
mode = mode 
up = up 
left = left
down = down 
right = right 

[br_debug_trace.zip](https://github.com/user-attachments/files/19823362/br_debug_trace.zip)

darthcloud, 2025-04-20:
This beta might fix it:

[v25.04-beta-41-g68cdd22b_hw1_genesis.zip](https://github.com/user-attachments/files/19823386/v25.04-beta-41-g68cdd22b_hw1_genesis.zip)

charlesmckee47, 2025-04-20:
testing as soon as it uploads to the blue retro

charlesmckee47, 2025-04-20:
same as before still broken in x input mode with button mapping looking like this 
game pad  alt [6 button mode]   
a=b
b=a 
c=z
z=x
y=x
x=y
start = start 
mode = mode 
up = up 
left = left
down = down 
right = right 

[br_debug_trace (2).zip](https://github.com/user-attachments/files/19823507/br_debug_trace.2.zip)

charlesmckee47, 2025-04-20:
should i maybe try another mode

charlesmckee47, 2025-04-20:
ok i just found out in android mode i can get the 3 main face button to behave right when changing a few config with the app  so 
A=a 
B=b
c=c 
but xyz are still messed up and i cant fix it with the advance config
x=z
z=x
y=y

charlesmckee47, 2025-04-20:
this is what i get in default config  in android mode i can config  it to were i can get c to work by changing z  but i cant get the to row fixed within advance config
game pad  alt [6 button mode]   
a=a
b=b
c=z
z=x
y=y
x=x
start = start 
mode = mode 
up = up 
left = left
down = down 
right = right game pad  alt [6 button mode]   
a=a
b=b
c=z
z=x
y=y
x=x
start = start 
mode = mode 
up = up 
left = left
down = down 
right = right 
[br_debug_trace (3).zip](https://github.com/user-attachments/files/19823568/br_debug_trace.3.zip)

charlesmckee47, 2025-04-20:
Well above me testing with Android mode is me testing with x input and it was still completely broken

charlesmckee47, 2025-04-20:
same as before still broken in x input mode with button mapping looking like this 
game pad  alt [6 button mode]   
a=b
b=a 
c=z
z=x
y=x
x=y
start = start 
mode = mode 
up = up 
left = left
down = down 
right = right 

[br_debug_trace.2.zip](https://github.com/user-attachments/files/19824664/br_debug_trace.2.zip)

darthcloud, 2025-04-21:
Try this one in X-input mode:

[v25.04-beta-44-g55e87da5_hw1_genesis.zip](https://github.com/user-attachments/files/19827847/v25.04-beta-44-g55e87da5_hw1_genesis.zip)

charlesmckee47, 2025-04-21:
nope i know it trial and error to get the result 
game pad  alt [6 button mode]   
a=b
b=a
c=z
z=c
y=x
x=y
start = start 
mode = mode 
up = up 
left = left
down = down 
right = right 

[br_debug_trace (4).zip](https://github.com/user-attachments/files/19828115/br_debug_trace.4.zip)

darthcloud, 2025-04-21:
ok  its hard to figure without the controller.

no need to note what the button makes on the genesis

just make another debug trace
connect controller 

and press the button exactly once in that order

start
mode
a
b
c
x
y
z
start
start

last two start is to help confirm the end

thanks

darthcloud, 2025-04-21:
just power off

charlesmckee47, 2025-04-21:
ok just finished the debug just waiting for my GameCube blue retro to finish updating and then i will send debug

charlesmckee47, 2025-04-21:
[br_debug_trace (5).zip](https://github.com/user-attachments/files/19828348/br_debug_trace.5.zip)

darthcloud, 2025-04-21:
Great thank you,

The problem with A,B,X,Y is that the 8bitdo controller is configured to invert them.

[image]

Try the HOME + DOWN option to change that.

I invrted C & Z so I'm making a new build to fix that one.

darthcloud, 2025-04-21:
[v25.04-beta-45-g3b2e104b_hw1_genesis.zip](https://github.com/user-attachments/files/19828560/v25.04-beta-45-g3b2e104b_hw1_genesis.zip)

This beta will fix C & Z, and using the 8bitdo HOME+DOWN hot key should fix A, B, X, Y

charlesmckee47, 2025-04-21:
I will check it once I get home tomorrow as I'm not wanting to return on my Genesis tonight

charlesmckee47, 2025-04-21:
Just checked and as long as you do what you said in the home and down with the latest beta that fixes the mod kit  I have just confirmed with my Genesis I did the button mapping shut the controller off turn the system on turn the controller on and then it worked thank you for your help and please continue to keep doing this amazing work for the  retro community if you would like I can give you a debug file

closed 2025-04-21 by darthcloud