darthcloud/BlueRetro #1219: 8bitdo ultimate 2 wireless controller bluetooth

INDRAPhilip, 2025-04-19:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

25.01

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

retroscaler

### System used

Sony PlayStation 2

### Bluetooth controller brand & name

BlueRetro e BlueRetro mini

### What is the problem? (only list ONE problem per report)

Control 8bitdo ultimate 2 wireless controller and bluetooth in firmware version 25.01 when connecting stay clicking alone preventing any means of play or proper operation

note: in the BlueRetro version with firmware 1.9 HW1 the controller was working without this problem in the latest firmware 24.10 in the BlueRetro mini connect, but without working even the buttons

### What did you expect to happen?

The controller works normally on all buttons, as it does with my other ps4 and ps5 controllers. 

### Attach BlueRetro debug trace here

how do i do that?

darthcloud, 2025-04-19:
This likely work in latest beta:
https://github.com/darthcloud/BlueRetro/actions/runs/14544498684

darthcloud, 2025-04-27:
Fixed in v25.04.

closed 2025-04-27 by darthcloud