darthcloud/BlueRetro #1234: [DC] Emulated VMU corrupts saves

pcercuei, 2025-05-10:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

25.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter with detachable cord

### Manufacturer

blue-retro.com

### System used

SEGA Dreamcast

### Bluetooth controller brand & name

Nintendo Switch Pro controller

### What is the problem? (only list ONE problem per report)

After a fresh format of the emulated VMU, copying back saves from a different VMU will corrupt the saves.
This seems to happen on my hardware after the first ~60 blocks are filled; anything else will either have partially corrupted files (e.g. the save file's icon looks different) or completely corrupted files (unrecognized by the BIOS).

### What did you expect to happen?

The emulated VMU should not corrupt the saves.

### Attach BlueRetro debug trace here

No debug trace as it requires the emulated memory to be disabled.

darthcloud, 2025-05-17:
I got a BR VMU almost full with 25 saves totaling 195 blocks and I can copy it to a real VMU and then copy the real VMU back to BR VMU fine with no corruption.

I did notice that if the BT controller disconnect while doing the copy than the transfer is aborted since BR disable the port whn their is no BT controller connected.

I made a couple change the prevent that from happening.

Can you try this beta and see if it work better for you?

[v25.07-beta-4-g7076d1b2_hw1_universal.zip](https://github.com/user-attachments/files/20261020/v25.07-beta-4-g7076d1b2_hw1_universal.zip)

darthcloud, 2025-05-20:
I spoke too soon sorry, while everything look fine while copying at the memery card screen once in game saves file located futher in the mem card are indeed reported as corrupted by games.

Your 60 blocks estimate sound correct. Looking into that this week.

pcercuei, 2025-05-20:
@darthcloud I was away the last few days so I could not test your beta yet. But I am glad you could reproduce it!

darthcloud, 2025-05-23:
This build should fix the corruption at the specific offset:
https://github.com/darthcloud/BlueRetro/actions/runs/15219805142

Basicly one of the memory allocation was done in an area from which the 2nd core couldn't write to.
That resulted in VMU block 0x98 though 0x9F to be unwritable in the latest build. But that was likely kind of random from build to build.

This build fix that by specifying to malloc that the memory need to be DMA accessible which in turn guarantee that its writable from the bare core.

I fixed a few others issue around VMU support too.

Some issue still remain with some games like Resident Evil 2 which look to be related to my maple interface driver RX.

darthcloud, 2025-05-25:
RE2 work with this one:
https://github.com/darthcloud/BlueRetro/actions/runs/15233625264

pcercuei, 2025-05-25:
It seems to be working much better with yesterday's build, thanks!
I'll play a little bit with it this week and report back.

naxeras, 2025-06-04:
> RE2 work with this one: https://github.com/darthcloud/BlueRetro/actions/runs/15233625264

It is possible to download this beta, Emulated VMU is not working in Kiss Psycho Circus

pcercuei, 2025-06-10:
@darthcloud It seems to work perfectly now, I didn't find any other issues. You can close the bug report.

darthcloud, 2025-06-10:
Thanks for testing!!

closed 2025-06-10 by darthcloud