darthcloud/BlueRetro #1238: HW2 Gamecube Laser Bear | Using the web config for Preset errors and doesn't load presets

typeiionair, 2025-05-18:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

v25.04

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

Laser Bear

### System used

Nintendo GameCube

### What is the problem? (only list ONE problem per report)

I receive the following output/error after connection on the Presets Config page. This error prevents the Mappings Config from loading the Presets. This worked once and I changed the preset and then after it wouldn't load. I've tried to FW reset & downgrade but nothing is solving the page error:

Reading App version...
Requesting Bluetooth Device...
Connecting to GATT Server...
Getting BlueRetro Service...
Init Cfg DOM...
Argh! SyntaxError: Unexpected token ']', ..." 0],
    ]
}" is not valid JSON

### What did you expect to happen?

The Preset options to load into the interface so that I can change it.

### Attach BlueRetro debug trace here

[br_debug_trace.zip](https://github.com/user-attachments/files/20274791/br_debug_trace.zip)

darthcloud, 2025-05-18:
Just fixed it, you may need to clear your browser cache or wait a bit.

closed 2025-05-18 by darthcloud

typeiionair, 2025-05-18:
Confirmed this resolved the issue.