darthcloud/BlueRetro #1243: [Feature request] Playstation 1 port status leds P3 P4

rutdarkronin, 2025-05-22:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

25.04

### BlueRetro firmware specification

HW2

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

Internal install DIY

### Manufacturer

ESP32-WROOM

### System used

Sony PlayStation 1

### Bluetooth controller brand & name

DualSense

### What is the problem? (only list ONE problem per report)

I have discovered that P3 and P4 cannot have their respective leds to light even though they're connected just fine. They're tied to IO12 and IO15, but there's also a notice that those pins are used to indicate analog mode on PSX - which is pretty useless if you ask me. Can we have an option of using this pins to actually indicate connected controllers? Or just move them to some other IOs?

### What did you expect to happen?

IO12 and IO15 outputs 3.3v when P3 and P4 connected to PSX.

rutdarkronin, 2025-05-22:
Sorry for that, no offense meant in any way.

darthcloud, 2025-05-22:
I'm not making change to the behavior of the LED since its basibly personal preference so its different for everybody.

People are already used to the current behavior.

The project is open source so you can modify the code to make your own version that match your taste.

closed 2025-05-22 by darthcloud