darthcloud/BlueRetro #1263: [HW1] sd2psXtd 1.2.0 playstation 1 issue

Kuslik, 2025-06-24:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

v25.04 hw1 playstation

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

DIY

### System used

Sony PlayStation 1

### Bluetooth controller brand & name

Sony Dualshock 3 CECHZC2R

### What is the problem? (only list ONE problem per report)

Psxmemcard pro (based on sd2psXtd 1.2.0) always throw an error or cant read memory card. For example, when my blueretro dongle and memcard both plugged Silent Hill can't proceed past "KCET" logo. Once blueretro is unplugged everything goes to normal. If continue to play like this saves don't work and game throws saving error, even if memory card is plugged to port 2

### What did you expect to happen?

To playststation being able to read and write memorycard

### Attach BlueRetro debug trace here

[br_debug_trace.zip](https://github.com/user-attachments/files/20893940/br_debug_trace.zip)

darthcloud, 2025-06-28:
Does regular memory card work fine on your PS1 while using your BlueRetro?

Did a previous version of BlueRetro worked, or it never did?

Kuslik, 2025-07-01:
Okay, so rolling back on sd2psx firware fixed it. I will create issues in their repo.

closed 2025-07-01 by Kuslik

Chandler-Kluser, 2025-11-29:
I face exactly same issue with recent blueretro firmware for my PU-18 playstation with picostation mod.

I flashed my [psxmemcard gen1](https://aliexpress.com/item/1005005032194366.html) `sd2psXtd` with [this firmware](https://github.com/sd2psXtd/firmware/releases/download/1.0.0/psxmemcard.uf2) (SHA256 `d9780a9e67a4b6be04786f470b0d1b1f1592c608238b784d9f4a12d425929bb2`) and my blueretro `25.04 hw1` for PSX and **both worked together** in internal mod with no TVS diodes, testing an [external adapter](https://aliexpress.com/item/1005004770273070.html) from [retro scaler](https://shoprenderview.aliexpress.com/credential/showcredential.htm) seemed to work but I did not made hard tests.