darthcloud/BlueRetro #1271: MemoryCard not detected on PS1

DANTE2501, 2025-07-05:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

25.04

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

Universal

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

RetroScaler

### System used

Sony PlayStation 1

### What is the problem? (only list ONE problem per report)

When using the adapter on a PlayStation 1 SCPH-5502 it prevents reading the MemoryCard on both ports. Tried different MemoryCards. and two PS1 Adapters. Also the cursor seems to be flashing. Tried the last firmware on date the 25.04 and also tried oldest ones but no chance. 
The adapter is not interfering with the MemoryCard when I use it on my PSOne SCPH-102. 

### What did you expect to happen?

I want to be able to use the Adapter along with my MemoryCards.

Kuslik, 2025-07-14:
I've experienced such troubles with simple blueretro dongle without [TVS diodes](https://github.com/darthcloud/BlueRetro/issues/102#issuecomment-1831116340). Building proper hardware fixed my issue.

Chandler-Kluser, 2025-08-21:
I believe he is not talking about getting random controller inputs but the impossibility of using a memory card in the same slot as the blueretro adapter.

I am talking about this because I have experienced such controller random inputs problem but the memory card issue still remains

DANTE2501, 2025-08-21:
> I believe he is not talking about getting random controller inputs but the impossibility of using a memory card in the same slot as the blueretro adapter.
> 
> I am talking about this because I have experienced such controller random inputs problem but the memory card issue still remains

DANTE2501, 2025-08-21:
> I believe he is not talking about getting random controller inputs but the impossibility of using a memory card in the same slot as the blueretro adapter.
> 
> I am talking about this because I have experienced such controller random inputs problem but the memory card issue still remains

Exactelly. And I was wondering if there is any chance that the issue could be solved in the future by a Firmware update

jlawless2690, 2025-11-26:
I had this issue I had to use an older firmware

jlawless2690, 2025-11-27:
it stopped my memory card slots to stop working, the controller worked fine with inputs and stuff, but it worked on PS2 witch was strange

Chandler-Kluser, 2025-11-27:
I mean what firmware version is in your adapter

DANTE2501, 2025-11-27:
> I had this issue I had to use an older firmware

Which firmware did solve your issue ?

Chandler-Kluser, 2025-11-29:
I have found [the solution](https://github.com/darthcloud/BlueRetro/issues/1263#issuecomment-3591801457)