darthcloud/BlueRetro #1276: Pairing with PC browser for config mostly fails

egparadigm, 2025-07-14:
### Answers checklist.

- [x] I have read and will follow the terms above.
- [x] I have read the documentation [BlueRetro wiki](https://github.com/darthcloud/BlueRetro/wiki) and the issue is not addressed there.
- [x] I have updated my BlueRetro to [latest version available](https://github.com/darthcloud/BlueRetro/releases/latest).
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### BlueRetro firmware version

25.07 beta

### BlueRetro firmware specification

HW1

### BlueRetro firmware variant

System specific

### BlueRetro hardware type

External adapter dongle (1 port only)

### Manufacturer

Blue Retro / Retro Scaler

### System used

Nintendo Wii Extension

### Bluetooth controller brand & name

NSO Gamecube

### What is the problem? (only list ONE problem per report)

I have three different Blue Retro dongles for PS1, N64 and GameCube (latter mostly used on Wii). I have the same difficulty with all three and the controller is not relevant, because it’s related to pairing to Windows in order to use the web config. 

I am using Windows 11 and Chrome or Edge, both of which exhibit the same problem and always have done despite firmware updates. Frequently (>50% of the time) the web config complains “argh” and the connection fails. I have to use some combination of closing the browser, powering down (disconnecting/reconnecting) the dongle, restarting Bluetooth and, sometimes, actually restarting Windows. There is no one combination of these actions that works reliably. It’s really quite annoying.

So I am hoping you can help me diagnose and fix the issue. I of course don’t know if it’s the dongles or my PC that’s causing the problem.

Thanks for an otherwise brilliant product and your attention.

### What did you expect to happen?

To connect seamlessly each time it is required.

darthcloud, 2025-07-14:
This was a problem mostly with FW lower than 24.10. It should be good in anything over that.

How to you update if the connection fail? You probably still got the old FW running?

Your best bet on these older FW is to to use a Windows base laptop with Chrome browser. Try multiple time, use a new Chrome tab for each try, try to change laptop position around BlueRetro etc.. If nothing work then try another laptop. If that doesn't work try to update it at a different location.

egparadigm, 2025-07-14:
Thanks for reply. Good to hear that this issue has improved mostly since 24.10. The device I am using today (GameCube on Wii) is on 25.07 beta and is still presenting the issue.

Although it doesn't seem so relevant, I have been able to update fw because sometimes connection succeeds.

I have tried about five times to download a debug trace but unfortunately it has failed every time, getting only part way to download on the progress bar.

Whilst I was doing this I wondered if it might be interference. I switched all my devices off. The only thing that seemed to make much difference was being in the same room at least.

BTW, the BR devices have often been active alongside Memcard Pro, at least for PS1/GC/Wii.